xpack-dev-tools/riscv-none-elf-gcc-xpack #9: build error on ubuntu20

dreamcmi, 2023-02-26:
Hello, I tried to compile the riscv toolchain of gcc12.2 on the Ubuntu20.04 system, and executed the relevant commands according to the instructions, but it seems that an error occurred, and no obvious error was found in the steps before prepare. When executing the build, an error that seems to be a permission problem occurred unexpectedly. I don’t know how to solve the following log output. I need your help, thank you
```shell
darren@~ $ git -C ~/Work/xpacks/riscv-none-elf-gcc-xpack.git pull && \
> xpm run install -C ~/Work/xpacks/riscv-none-elf-gcc-xpack.git && \
> git -C ~/Work/xpacks/xbb-helper-xpack.git pull && \
> xpm link -C ~/Work/xpacks/xbb-helper-xpack.git && \
> xpm run link-deps -C ~/Work/xpacks/riscv-none-elf-gcc-xpack.git && \
> xpm run deep-clean --config linux-x64 -C ~/Work/xpacks/riscv-none-elf-gcc-xpack.git && \
> xpm run docker-prepare --config linux-x64 -C ~/Work/xpacks/riscv-none-elf-gcc-xpack.git && \
> xpm run docker-link-deps --config linux-x64 -C ~/Work/xpacks/riscv-none-elf-gcc-xpack.git
Already up to date.
> npm install

up to date, audited 3 packages in 2s

1 package is looking for funding
  run `npm fund` for details

found 0 vulnerabilities
> xpm install
@xpack-dev-tools/riscv-none-elf-gcc...
+ @xpack-dev-tools/xbb-helper@1.4.7
'xpacks/xpack-dev-tools-xbb-helper' -> '/home/darren/.local/xPacks/@xpack-dev-tools/xbb-helper/1.4.7'
Already up to date.
@xpack-dev-tools/xbb-helper -> '/home/darren/Work/xpacks/xbb-helper-xpack.git'
> xpm link @xpack-dev-tools/xbb-helper
@xpack-dev-tools/xbb-helper -> '/home/darren/Work/xpacks/xbb-helper-xpack.git'
> rm -rf build/linux-x64
> rm -rf ${HOME}/Work/riscv-none-elf-gcc-[0-9]*-*/linux-x64
> mkdir -pv ${HOME}/.wine
> docker rm --force riscv-none-elf-gcc-12.2.0-3.1-linux-x64
Error: No such container: riscv-none-elf-gcc-12.2.0-3.1-linux-x64
> docker create --name riscv-none-elf-gcc-12.2.0-3.1-linux-x64 --tty --hostname docker --volume $(pwd):$(pwd) --volume ${HOME}/Work/xpacks/xbb-helper-xpack.git:${HOME}/Work/xpacks/xbb-helper-xpack.git --volume ${HOME}/.local/xPacks:${HOME}/.local/xPacks --volume ${HOME}/.cache/xPacks:${HOME}/.cache/xPacks --volume ${HOME}/Work/cache:${HOME}/Work/cache --volume ${HOME}/.wine:${HOME}/.wine --workdir $(pwd) ilegeul/ubuntu:amd64-18.04-xbb-v5.0.0
b05199aa65ad38046fbac7a6ab3aa96cc94ae3ba0826fbebd1de2e8566b5a10a
> docker start riscv-none-elf-gcc-12.2.0-3.1-linux-x64
riscv-none-elf-gcc-12.2.0-3.1-linux-x64
> docker exec riscv-none-elf-gcc-12.2.0-3.1-linux-x64 npm install --location=global xpm@latest

added 2 packages, and audited 492 packages in 38s

28 packages are looking for funding
  run `npm fund` for details

7 vulnerabilities (6 moderate, 1 high)

To address issues that do not require attention, run:
  npm audit fix

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.
npm notice
npm notice New major version of npm available! 8.19.2 -> 9.5.1
npm notice Changelog: <https://github.com/npm/cli/releases/tag/v9.5.1>
npm notice Run `npm install -g npm@9.5.1` to update!
npm notice
> docker exec riscv-none-elf-gcc-12.2.0-3.1-linux-x64 userdel node
> docker exec riscv-none-elf-gcc-12.2.0-3.1-linux-x64 groupadd --gid $(id -g) --force $(id -gn)
> docker exec riscv-none-elf-gcc-12.2.0-3.1-linux-x64 useradd --home-dir ${HOME} --uid $(id -u) --gid $(id -g) $(id -un)
> docker exec --user $(id -un) riscv-none-elf-gcc-12.2.0-3.1-linux-x64 bash -c 'lsb_release -sd && whoami && pwd && ls -lLA'
Ubuntu 18.04.6 LTS
darren
/home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git
total 120
drwxrwxr-x 8 darren darren  4096 Feb 26 14:09 .git
drwxrwxr-x 3 darren darren  4096 Feb 26 13:41 .github
-rw-rw-r-- 1 darren darren  1732 Feb 26 13:41 .gitignore
-rw-rw-r-- 1 darren darren     0 Feb 26 13:41 .gitmodules
-rw-rw-r-- 1 darren darren   130 Feb 26 13:41 .npmignore
[... 400 lines cut ...]
Here we go...

mkdir: created directory '/home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git/build/linux-x64/sources'

Downloading "gcc-12.2.0.tar.xz" from "https://ftp.gnu.org/gnu/gcc/gcc-12.2.0/gcc-12.2.0.tar.xz"...

[curl --insecure --fail --location --output /home/darren/Work/cache/gcc-12.2.0.tar.xz.download https://ftp.gnu.org/gnu/gcc/gcc-12.2.0/gcc-12.2.0.tar.xz]
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0Warning: Failed to create the file
Warning: /home/darren/Work/cache/gcc-12.2.0.tar.xz.download: Permission denied
  0 80.7M    0 16384    0     0   3239      0  7:15:33  0:00:05  7:15:28  4248
curl: (23) Failed writing body (0 != 16384)

Make the build folder writable by all...

[chmod -R a+w /home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git/build]

Make the xpacks folder writable by all...

[chmod a+w /home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git/xpacks]
error: running 'bash  scripts/build.sh --target linux-x64 --build-folder build/linux-x64 --develop' failed
error: running 'docker exec --user $(id -un) riscv-none-elf-gcc-12.2.0-3.1-linux-x64 xpm run build-develop --config linux-x64' failed
darren@~ $
```

ilg-ul, 2023-02-26:
It looks like a problem with the `/home/darren/Work/cache` folder.

Could you rename or remove it and retry?

dreamcmi, 2023-02-26:
Thanks, I checked the following cache directory and found that it is actually under the root user group. It should be that a wrong command was executed during the prepare phase, resulting in a user group assignment error. I have changed the cache user group and is compiling. If there is no related Error, I will close this issue.

dreamcmi, 2023-02-26:
ops, It made an error again. This time, cmake could not be found. I checked that my host machine has cmake. Is the docker image not installed?
```shell
Running zstd cmake...

[cmake -LAH -G Ninja -DCMAKE_INSTALL_PREFIX=/home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git/build/linux-x64/x86_64-pc-linux-gnu/install -DZSTD_BUILD_CONTRIB=ON -DZSTD_BUILD_CONTRIB=OFF -DZSTD_BUILD_PROGRAMS=OFF -DZSTD_BUILD_STATIC=ON -DZSTD_LEGACY_SUPPORT=ON -DZSTD_ZLIB_SUPPORT=ON -DZSTD_LZMA_SUPPORT=ON -DZSTD_PROGRAMS_LINK_SHARED=ON /home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git/build/linux-x64/sources/zstd-1.5.2/build/cmake]
/home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git/xpacks/xpack-dev-tools-xbb-helper/scripts/wrappers.sh: line 34: cmake: command not found

Make the build folder writable by all...

[chmod -R a+w /home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git/build]

Make the xpacks folder writable by all...

[chmod a+w /home/darren/Work/xpacks/riscv-none-elf-gcc-xpack.git/xpacks]
error: running 'bash  scripts/build.sh --target linux-x64 --build-folder build/linux-x64 --develop' failed
error: running 'docker exec --user $(id -un) riscv-none-elf-gcc-12.2.0-3.1-linux-x64 xpm run build-develop --config linux-x64' failed
darren@~/Work $ cmake --version
cmake version 3.16.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).

```

ilg-ul, 2023-02-26:
You are either using an older `xpm`, or invoking the build with the wrong command, since `cmake` was installed by `xpm` in the docker image (as per your listing above).

dreamcmi, 2023-02-26:
I have checked the `xpm` verison is `0.14.9`. The reason should be that my network conditions are too poor and I have not downloaded all dependencies. Is there a way to manually download dependencies?

```shell
> docker exec --user $(id -un) riscv-none-elf-gcc-12.2.0-3.1-linux-x64 xpm install
@xpack-dev-tools/riscv-none-elf-gcc...
+ @xpack-dev-tools/xbb-helper@1.4.7
'xpacks/xpack-dev-tools-xbb-helper' -> '/home/darren/.local/xPacks/@xpack-dev-tools/xbb-helper/1.4.7'
> docker exec --user $(id -un) riscv-none-elf-gcc-12.2.0-3.1-linux-x64 xpm install --config linux-x64
@xpack-dev-tools/riscv-none-elf-gcc --config linux-x64...
+ @xpack-dev-tools/patchelf@0.17.2-1.1
+ @xpack-dev-tools/gcc@12.2.0-2.1
+ @xpack-dev-tools/m4@1.4.19-2.1
+ @xpack-dev-tools/pkg-config@0.29.2-2.1
+ @xpack-dev-tools/ninja-build@1.11.1-2.1
+ @xpack-dev-tools/cmake@3.23.5-1.1
@xpack-dev-tools/pkg-config@0.29.2-2.1 => '/home/darren/.local/xPacks/@xpack-dev-tools/pkg-config/0.29.2-2.1'
@xpack-dev-tools/cmake@3.23.5-1.1 => '/home/darren/.local/xPacks/@xpack-dev-tools/cmake/3.23.5-1.1'
@xpack-dev-tools/patchelf@0.17.2-1.1 => '/home/darren/.local/xPacks/@xpack-dev-tools/patchelf/0.17.2-1.1'
@xpack-dev-tools/m4@1.4.19-2.1 => '/home/darren/.local/xPacks/@xpack-dev-tools/m4/1.4.19-2.1'
@xpack-dev-tools/gcc@12.2.0-2.1 => '/home/darren/.local/xPacks/@xpack-dev-tools/gcc/12.2.0-2.1'
@xpack-dev-tools/ninja-build@1.11.1-2.1 => '/home/darren/.local/xPacks/@xpack-dev-tools/ninja-build/1.11.1-2.1'
Downloading https://github.com/xpack-dev-tools/pkg-config-xpack/releases/download/v0.29.2-2/xpack-pkg-config-0.29.2-2-linux-x64.tar.gz...
Downloading https://github.com/xpack-dev-tools/cmake-xpack/releases/download/v3.23.5-1/xpack-cmake-3.23.5-1-linux-x64.tar.gz...
Downloading https://github.com/xpack-dev-tools/m4-xpack/releases/download/v1.4.19-2/xpack-m4-1.4.19-2-linux-x64.tar.gz...
Downloading https://github.com/xpack-dev-tools/patchelf-xpack/releases/download/v0.17.2-1/xpack-patchelf-0.17.2-1-linux-x64.tar.gz...
Downloading https://github.com/xpack-dev-tools/gcc-xpack/releases/download/v12.2.0-2/xpack-gcc-12.2.0-2-linux-x64.tar.gz...
Downloading https://github.com/xpack-dev-tools/ninja-build-xpack/releases/download/v1.11.1-2/xpack-ninja-build-1.11.1-2-linux-x64.tar.gz...
request to https://github.com/xpack-dev-tools/cmake-xpack/releases/download/v3.23.5-1/xpack-cmake-3.23.5-1-linux-x64.tar.gz failed, reason: read ECONNRESET in fetch https://github.com/xpack-dev-tools/cmake-xpack/releases/download/v3.23.5-1/xpack-cmake-3.23.5-1-linux-x64.tar.gz
error: download failed, quit
error: running 'docker exec --user $(id -un) riscv-none-elf-gcc-12.2.0-3.1-linux-x64 xpm install --config linux-x64' failed
```

ilg-ul, 2023-02-26:
> my network conditions are too poor and I have not downloaded all dependencies.

That's unfortunate, but, with some patience, can be overcome, since the build process uses caches, and once the archives downloaded, are used without any download.

> Is there a way to manually download dependencies?

Yes, remove the dependency folder in `~/.local/xPacks` and issue `xpm install --global <dependency>` until it succeeds. This downloads the dependency into the cache and extracts it in the central store, without creating links to the project.

Later, when the project tries to install the dependency, it should be already there.

For the archives required during the build, as you noticed, the files are downloaded into `~/Work/cache`. If the download fails, restart the script until the download succeeds.

dreamcmi, 2023-02-27:
I found a problem. I have set up the system proxy (http_proxy https_proxy), and wget can access the proxy normally, but `xpm` does not use the system proxy. Is this the design or is there a mistake in my settings?
```shell
darren@darren-virtual-machine:~$ xpm install --global @xpack-dev-tools/gcc@12.2.0-2.1 --force
@xpack-dev-tools/gcc@12.2.0-2.1...
@xpack-dev-tools/gcc@12.2.0-2.1 => '/home/darren/.local/xPacks/@xpack-dev-tools/gcc/12.2.0-2.1'
Downloading https://github.com/xpack-dev-tools/gcc-xpack/releases/download/v12.2.0-2/xpack-gcc-12.2.0-2-linux-x64.tar.gz...
request to https://github.com/xpack-dev-tools/gcc-xpack/releases/download/v12.2.0-2/xpack-gcc-12.2.0-2-linux-x64.tar.gz failed, reason: read ECONNRESET in fetch https://github.com/xpack-dev-tools/gcc-xpack/releases/download/v12.2.0-2/xpack-gcc-12.2.0-2-linux-x64.tar.gz
error: download failed, quit
darren@darren-virtual-machine:~$ wget https://github.com/xpack-dev-tools/gcc-xpack/releases/download/v12.2.0-2/xpack-gcc-12.2.0-2-linux-x64.tar.gz
--2023-02-27 12:43:02--  https://github.com/xpack-dev-tools/gcc-xpack/releases/download/v12.2.0-2/xpack-gcc-12.2.0-2-linux-x64.tar.gz
Connecting to 192.168.20.100:7890... connected.
Proxy request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/256410491/0ad10e2d-e752-4c2f-a44d-91bcdc16b99b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230227%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230227T044303Z&X-Amz-Expires=300&X-Amz-Signature=04ab952870f2ed843fc06b6993707444c3c40fa4eb7a923577ac3bc998f7f72d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=256410491&response-content-disposition=attachment%3B%20filename%3Dxpack-gcc-12.2.0-2-linux-x64.tar.gz&response-content-type=application%2Foctet-stream [following]
--2023-02-27 12:43:03--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/256410491/0ad10e2d-e752-4c2f-a44d-91bcdc16b99b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230227%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230227T044303Z&X-Amz-Expires=300&X-Amz-Signature=04ab952870f2ed843fc06b6993707444c3c40fa4eb7a923577ac3bc998f7f72d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=256410491&response-content-disposition=attachment%3B%20filename%3Dxpack-gcc-12.2.0-2-linux-x64.tar.gz&response-content-type=application%2Foctet-stream
Connecting to 192.168.20.100:7890... connected.
Proxy request sent, awaiting response... 200 OK
Length: 137017840 (131M) [application/octet-stream]
Saving to: ‘xpack-gcc-12.2.0-2-linux-x64.tar.gz.2’

xpack-gcc-12.2.0-2-linux-x64.tar.gz.2   100%[=============================================================================>] 130.67M  21.0MB/s    in 9.2s    

2023-02-27 12:43:15 (14.3 MB/s) - ‘xpack-gcc-12.2.0-2-linux-x64.tar.gz.2’ saved [137017840/137017840]

```

ilg-ul, 2023-02-27:
Ah, sorry, I did not have the time to investigate how to download the archives using a proxy in xpm. :-(

For downloads I use [node-fetch](https://www.npmjs.com/package/node-fetch), which is a classical, but does not mention anything about proxies.

Personally I never used a proxy, and have no experience with such configurations. Could you provide more details about your configuration, how did you make it work in wget, and what would be a possible solution for xpm?

To be noted that in xpm there are two separate downloads, the first is the npm package itself, downloaded from npmjs.com, which seems to pass through your proxy, and then the binary archive, which fails.

dreamcmi, 2023-02-27:
I temporarily solved the problem by putting the proxy into the router and it compiled successfully.
For the problem of proxy settings, the following command was executed
```shell
export http_proxy=http://192.168.20.100:7890
export https_proxy=http://192.168.20.100:7890
```
I'm not familiar with JavaScript, but I found node-fetch proxy method. Like this.
```javascript
fetch(url, { 
    method: 'GET',
    agent: new HttpProxyAgent("http://" + ip + ":" + port) 
})
```
I wrote a piece of pseudocode showing my approach.
```c
void test(String url)
{
    String proxy_string = get_environment("http_proxy");
    if (len(proxy_string) > 0)
    {
        fetch(url, {method: 'GET', agent: new HttpProxyAgent(proxy_string)})
    }
    else
    {
        fetch(url)
    }
}
```

ilg-ul, 2023-02-27:
I found an old issue that was closed prematurely by the author:

- https://github.com/xpack/xpm-js/issues/80#issuecomment-773776574

I re-opened it, and will try to fix it in the next xpm release.

dreamcmi, 2023-02-27:
That's greate! Thanks.

closed 2023-02-27 by dreamcmi

ilg-ul, 2023-02-27:
Can you confirm that in your configuration the proxy definitions are only the two environment variables and you did not configure a proxy in the npm config files?
 
I think I identified how to forward the fetch() requests via a proxy (using the agent you mentioned). Now I need to identify where the definitions come from.

dreamcmi, 2023-02-28:
I did not set a proxy for npm, but directly switched to a domestic mirror source.

ilg-ul, 2023-02-28:
Since in your initial run, getting the npm package was functional, and only getting the binary archive failed, there must be a proxy setting somewhere, either in the npm configuration files or in the environment.

- do you have any npm configuration for the proxy? (`npm config ls -l | grep -i proxy`)
- do you have any environment variables with the proxy definitions? (`env | grep -i proxy`)

dreamcmi, 2023-02-28:
`do you have any npm configuration for the proxy? (npm config ls -l | grep -i proxy)`
```shell
darren@darren-virtual-machine:~/Desktop$ npm config ls -l | grep -i proxy
https-proxy = null 
noproxy = [""] 
proxy = null 
```
Although I have not set up the npm proxy, but xpm can check this part. For users who only set up the npm proxy, the relationship between the npm proxy and the global proxy should be or.

`do you have any environment variables with the proxy definitions? (env | grep -i proxy)`
```shell
darren@darren-virtual-machine:~/Desktop$ env | grep -i proxy
no_proxy=localhost,127.0.0.0/8,::1
https_proxy=http://192.168.20.100:7890/
NO_PROXY=localhost,127.0.0.0/8,::1
HTTPS_PROXY=http://192.168.20.100:7890/
HTTP_PROXY=http://192.168.20.100:7890/
http_proxy=http://192.168.20.100:7890/
ALL_PROXY=socks://192.168.20.100:7890/
all_proxy=socks://192.168.20.100:7890/
```
The capitalization problem here is due to the settings of Ubuntu desktop. In fact, you only need to care about the lowercase part (http_proxy https_proxy), because the command line version system settings only need lowercase.

ilg-ul, 2023-02-28:
Ok, there are no npm configs related to the proxy, and there are several environment variables. 

As for the upper/lower case, I have to further investigate, but the general rule is that environment variables use only upper case letters. I have no idea why in your environment both upper and lower case names are defined. 

Can you check which files define those proxy environment variables? Are both upper and lower cases definitions in the same file or in different files (all lower case in on file and all upper case in another file)

dreamcmi, 2023-02-28:
The Ubuntu desktop version uses `gsetting` to set the proxy. But I don't know where they are in the file system.
```shell
darren@darren-virtual-machine:~$ gsettings list-schemas | grep proxy
org.gnome.system.proxy
org.gnome.system.proxy.ftp
org.gnome.system.proxy.http
org.gnome.system.proxy.https
org.gnome.system.proxy.socks
darren@darren-virtual-machine:~$ gsettings get org.gnome.system.proxy.http host
'192.168.20.100'
darren@darren-virtual-machine:~$ gsettings get org.gnome.system.proxy.http port
7890
```
But the common solution is to edit `~/.bashrc` and add the following statement. Then `source ~/.bashrc`.
```shell
export http_proxy=http://192.168.20.100:7890
export https_proxy=http://192.168.20.100:7890
```