xpack-dev-tools/riscv-none-elf-gcc-xpack #10: "How to build" link in documentation is broken

TommyMurphyTM1234, 2023-02-27:
On this page:

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/

The "How to build" link:

* https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/xpack/README-BUILD.md

is broken and gives a 404 error.

TommyMurphyTM1234, 2023-02-27:
Looks like the same issue applies to all 3rd party and other tools too:

* https://xpack.github.io/dev-tools/aarch64-none-elf-gcc/#build-details
  * https://github.com/xpack-dev-tools/aarch64-none-elf-gcc-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/arm-none-eabi-gcc/#build-details
  * https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/clang/#build-details
  * https://github.com/xpack-dev-tools/clang-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/cmake/#build-details
  * https://github.com/xpack-dev-tools/cmake-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/gcc/#build-details
  * https://github.com/xpack-dev-tools/gcc-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/meson-build/#build-details
  * https://github.com/xpack-dev-tools/meson-build-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/mingw-w64-gcc/#build-details
  * https://github.com/xpack-dev-tools/mingw-w64-gcc-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/ninja-build/#build-details
  * https://github.com/xpack-dev-tools/ninja-build-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/openocd/#build-details
  * https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/qemu-arm/#build-details
  * https://github.com/xpack-dev-tools/qemu-arm-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/qemu-riscv/#build-details
  * https://github.com/xpack-dev-tools/qemu-riscv-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#build-details
  * https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/windows-build-tools/#build-details
  * https://github.com/xpack-dev-tools/windows-build-tools-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/m4/#build-details
  * https://github.com/xpack-dev-tools/m4-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/patchelf/#build-details
  * https://github.com/xpack-dev-tools/patchelf-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/pkg-config/#build-details
  * https://github.com/xpack-dev-tools/pkg-config-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/realpath/#build-details
  * https://github.com/xpack-dev-tools/realpath-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/sed/#build-details
  * https://github.com/xpack-dev-tools/sed-xpack/blob/xpack/README-BUILD.md
* https://xpack.github.io/dev-tools/wine/#build-details
  * https://github.com/xpack-dev-tools/wine-xpack/blob/xpack/README-BUILD.md

ilg-ul, 2023-02-27:
Thank you, Tommy, for spotting this, you are right, those links are broken. 

The fix is on the way. 

In the future, if you identify such problems affecting all projects, you don't have to provide the complete list of links, since I can easily do a global find & replace in the web sources.

TommyMurphyTM1234, 2023-06-01:
No longer an issue - thanks.

closed 2023-06-01 by TommyMurphyTM1234