xpack-dev-tools/riscv-none-elf-gcc-xpack #17: rv32ec_zicsr tries to link against RVI libgcc

jnk0le, 2023-05-20:
whenever `-march=rv32ec_zicsr -mabi=ilp32e` is used any reference to standard lib (newlib nano specs) ends up with:
```
c:/program files/gnu mcu eclipse/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: error: c:/program files/gnu mcu eclipse/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-memcpy-asm.o): mis-matched ISA string to merge 'i' and 'e'
c:/program files/gnu mcu eclipse/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: failed to merge target specific data of file c:/program files/gnu mcu eclipse/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-memcpy-asm.o)
```
The only working (with csr insns) workaround is to use `-march=rv32ec -mabi=ilp32e -misa-spec=2.2` (https://www.eevblog.com/forum/microcontrollers/risc-v-compilation-errors-mismatched-libgcc/)

ilg-ul, 2023-05-20:
Thank you for the link, but can you summarise the issue?

jnk0le, 2023-05-20:
The list doesn't contain _zicsr extension, but Zicsr ext will be useless here and I belive that it should't be added to the permutation set.

In this case compiler should pull in the lowest common denominator being rv32ec.
Instead it's pullnig RVI one (rv32i2p1_m2p0_a2p1_c2p0 ?)

here is log when trying `printf()`
```
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: error: c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-errno.o) use 16-byte stack aligned but the output use 4-byte stack aligned
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: error: c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-errno.o): mis-matched ISA string to merge 'i' and 'e'
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: failed to merge target specific data of file c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-errno.o)
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: error: c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-impure.o) use 16-byte stack aligned but the output use 4-byte stack aligned
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: rv32i2p1_m2p0_a2p1_c2p0: standard ISA extension `i' is not in canonical order
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: failed to merge target specific data of file c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-impure.o)
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: error: c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-printf.o) use 16-byte stack aligned but the output use 4-byte stack aligned
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: rv32i2p1_m2p0_a2p1_c2p0: standard ISA extension `i' is not in canonical order
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: failed to merge target specific data of file c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-printf.o)
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: error: c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-findfp.o) use 16-byte stack aligned but the output use 4-byte stack aligned
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: rv32i2p1_m2p0_a2p1_c2p0: standard ISA extension `i' is not in canonical order
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: failed to merge target specific data of file c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-findfp.o)
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: error: c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-fwalk.o) use 16-byte stack aligned but the output use 4-byte stack aligned
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: rv32i2p1_m2p0_a2p1_c2p0: standard ISA extension `i' is not in canonical order
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: failed to merge target specific data of file c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-fwalk.o)
[... cut ...]
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: rv32i2p1_m2p0_a2p1_c2p0: standard ISA extension `i' is not in canonical order
c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld.exe: failed to merge target specific data of file c:/program files/gnu mcu eclipse/gcc/xpack-riscv-none-elf-gcc-12.2.0-3/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib\libg_nano.a(lib_a-nano-msizer.o)
```

ilg-ul, 2023-05-20:
Sorry but I'm no longer involved in the RISC-V development, and I don't know which are the valid combinations, I use only common 32/64 build for my tests.

The current list was inherited from previous releases, without a thorough analyse.

If you know of a distribution that has a more up to date list, I can make a new release, but otherwise I don't know how to fix your problem. :-(

jnk0le, 2023-05-20:
did more tests and it seems that when there is no match for march string then it defaults to RV32GC libc

e.g. RV32I_Zba pulls in RV32GC, RV32IC_Zba gives a false positive of compiling without any issues until A or M instruction is hit.

>If you know of a distribution that has a more up to date list, I can make a new release, but otherwise I don't know how to fix your problem

increasing permutation set for useless things like `zicsr`, `zifencei` etc. is not the cleanest solution.
I'll check what upstream says.

ilg-ul, 2023-05-20:
> I'll check what upstream says.

Please note that the xPack distribution uses the upstream sources, since SiFive no longer makes their toolchain public.

ilg-ul, 2023-05-20:
@TommyMurphyTM1234, any comments on this issue?

TommyMurphyTM1234, 2023-05-20:
> @TommyMurphyTM1234, any comments on this issue?

Sorry, not at the moment. I'll need to try to recap on the latest state of play with regard to multilib generation, reuse, and canonicalised versus non-canonicalised architecture strings. All of which have caused me no end of confusion and which cause problems with building and/or using the toolchain lately.

TommyMurphyTM1234, 2023-06-01:
> Sorry, not at the moment. I'll need to try to recap on the latest state of play with regard to multilib generation, reuse, and canonicalised versus non-canonicalised architecture strings. All of [...]

See my comments here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1255#issuecomment-1563575921

TommyMurphyTM1234, 2023-06-01:
This issue/discussion might also be relevant here.

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1262

ilg-ul, 2023-06-01:
This is a bit too confusing for me. :-(

Do I have to adjust something in the multilib definition in my scripts?

TommyMurphyTM1234, 2023-06-01:
> This is a bit too confusing for me. :-(

For me too! :-|

> Do I have to adjust something in the multilib definition in my scripts?

I'm not sure.
These are my observations:

1. You might want to - and might already plan to - reflect the separation of the `Zicsr` and `Zifencei` extensions from the base integer ISA in the project settings GUI?
2. You might want to review [your choice of toolchain default `arch/abi` of `rv32imac/ilp32`](https://xpack.github.io/blog/2023/02/05/riscv-none-elf-gcc-v12-2-0-3-released/#supported-libraries) especially since it does not include `zicsr`?
3. If the end user selects an `arch/abi` for which there are no multilibs/startup code then I don't think that you can do much about that and any concomitant link or even runtime errors when the toolchain default `arch/abi` libs/startup code are linked instead? 
4. When `riscv-gnu-toolchain` bumps GCC from 12 to 13 (there's a PR still waiting acceptance) then the latter will include changes to how multilibs are matched. I think it's simply an automatic inference and use of the "multilib sharing" that users could do manually via `multilib-generator` but I'm still not completely clear on what it all involves or addresses. It may mean that it is possible to generate a smaller than exhaustive list of multilibs albeit at the cost of some programs being linked with multilibs/startup code that may not take full advantage of all available instructions.
5. As you already know, it's not really practical to generate multilibs for all possible combinations of extensions so some decision has to be taken as to where to draw the line. The previous point about the GCC 13 multilib handling/matching may be relevant here too. But, again, I still don't fully understand these changes.

TommyMurphyTM1234, 2023-06-01:
And we haven't even touched on proprietary extensions here yet! :-D
For example:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1261

TommyMurphyTM1234, 2023-06-01:
> You mean to make it include Zicsr?

Yes - on the basis that, as far as I know and have experienced, most or all RISC-V targets will require CSR access instructions in practice. But, as ever, there is no "correct" answer here.

> Now they are (in application.sh):
> 
> ```
> XBB_APPLICATION_GCC_ARCH=${XBB_APPLICATION_GCC_ARCH:-"rv32imac"}
> XBB_APPLICATION_GCC_ABI=${XBB_APPLICATION_GCC_ABI:-"ilp32"}
> ```
> 
> How would those two options look like?

Presumably `rv32imac` would become `rv32imac_zicsr`?

> As for the long list of multiIibs, if you know how to make them accept more useful cases without increasing the length of the list, I can consider it.

I *think* that the GCC 13 changes mean that where a multilib for a specific `arch/abi` is not available then the toolchain will instead select the "nearest" compatible one - but I am still not certain of the details. 

> For example why not add Zicsr and Zifencei to all libraries?

I don't know the answer to this question.
But for backward compatibility to when the `I` integer base ISA subsumed `Zicsr` and `Zifencei` perhaps that would make sense.
But then (unless GCC 13 changes things) if somebody selected `rv32ima/ilp32` but this multilib does not exist (because the toolchain instead bundles `rv32ima_zicsr_zifencei/ilp32`) then the toolchain default libs for `rv32imac[_zicsr_zifencei?]/ilp32` will be linked instead and the resulting executable may contain compressed instructions which will obviously trap on a target that does not actually implement the `C` extension.

jnk0le, 2024-11-17:
I think it should be already fixed (at least in gcc 13/14)

closed 2024-11-17 by jnk0le