xpack-dev-tools/riscv-none-elf-gcc-xpack #18: Release v10.2.0-1.2.1 not available through XPM

pur-jus, 2023-05-29:
### Description

No release v10.2.0-1.2.1 is available through XPM if this version is referenced with _package.json_

### Steps to Reproduce

package.json:

```json
    "devDependencies": {
      "@xpack-dev-tools/meson-build": "0.63.3-1.1",
      "@xpack-dev-tools/ninja-build": "1.11.1-2.1",
      "@xpack-dev-tools/riscv-none-elf-gcc": "10.2.0-1.2.1"
    },
```

Executing task: xpm install 

@my-env/my-template-project...
+ @xpack-dev-tools/meson-build@0.63.3-1.1
+ @xpack-dev-tools/ninja-build@1.11.1-2.1
error: Package '@xpack-dev-tools/riscv-none-elf-gcc@10.2.0-1.2.1' not found

**Expected behaviour:**

I would like to use this specific release.

**Actual behaviour:**

It is not possible to use this specific release.

Could you build this release?

ilg-ul, 2023-05-29:
Unfortunately 10.2 is a bit too old to be worth a rebuild. :-(

It was part of the now deprecated `riscv-none-embed-gcc` project:

- https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases

I guess you can still use it, but I don't recommend it.

pur-jus, 2023-05-29:
Thank you for your response. I use it because my project uses some old instructions like `csrrs`.

It's not possible to use XPM and VS Code extension with this old version?

ilg-ul, 2023-05-29:
As far as I remember, it is a valid package, so it should work with the current **xpm**, but no guarantees and no support.

You should consider updating your projects.

closed 2023-05-30 by pur-jus

ilg-ul, 2023-05-30:
Does the deprecated package still work?

pur-jus, 2023-06-01:
> Does the deprecated package still work?

Yes 👍 

```
xpm install --global @xpack-dev-tools/riscv-none-embed-gcc@10.2.0-1.2.1
```