xpack-dev-tools/riscv-none-elf-gcc-xpack #22: Linux x64 riscv-none-elf-gdb is built without ELF support

zador-blood-stained, 2023-09-19:
Linux x64 `riscv-none-elf-gdb` fails to load symbols from an ELF file.

Release 13.2.0-1 Linux x64:
```
➜  test-gdb  % riscv-none-elf-gdb --version
GNU gdb (xPack GNU RISC-V Embedded GCC x86_64) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
➜  test-gdb  % riscv-none-elf-gdb -q bme_sensor_oled.elf
Reading symbols from bme_sensor_oled.elf...
I'm sorry, Dave, I can't do that.  Symbol format `elf32-littleriscv' unknown.
(gdb) q
➜  test-gdb  %
```

Windows version works fine
```
X:\>xpack-riscv-none-elf-gcc-13.2.0-1\bin\riscv-none-elf-gdb.exe --version
GNU gdb (xPack GNU RISC-V Embedded GCC x86_64) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

X:\>xpack-riscv-none-elf-gcc-13.2.0-1\bin\riscv-none-elf-gdb.exe -q bme_sensor_oled.elf
Reading symbols from bme_sensor_oled.elf...
(gdb) q

X:\>
```

Older 12.3.0-1 release doesn't work (GDB version there is also 13.2)
```
➜  test-gdb  % xpack-riscv-none-elf-gcc-12.3.0-1/bin/riscv-none-elf-gdb --version
GNU gdb (xPack GNU RISC-V Embedded GCC x86_64) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
➜  test-gdb  % xpack-riscv-none-elf-gcc-12.3.0-1/bin/riscv-none-elf-gdb -q bme_sensor_oled.elf
Reading symbols from bme_sensor_oled.elf...
I'm sorry, Dave, I can't do that.  Symbol format `elf32-littleriscv' unknown.
(gdb) q
➜  test-gdb  %
```

11.3.0-1 release works fine (GDB version 12.1):
```
➜  test-gdb  % xpack-riscv-none-elf-gcc-11.3.0-1/bin/riscv-none-elf-gdb --version
GNU gdb (xPack GNU RISC-V Embedded GCC x86_64) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
➜  test-gdb  % xpack-riscv-none-elf-gcc-11.3.0-1/bin/riscv-none-elf-gdb -q bme_sensor_oled.elf
Reading symbols from bme_sensor_oled.elf...
(gdb) q
➜  test-gdb  %
```

Linux x64 `riscv-none-embed-gdb` (release 10.2.0-1.2) also works fine.

I checked build logs in GH Actions [here](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/actions/runs/5964910866) and I believe that this failing check is responsible for the ELF format support based on info from [this](https://sourceware.org/bugzilla/show_bug.cgi?id=26949) old bug report:
```
2023-08-24T16:44:57.6812604Z mkdir -p -- ./gdb
2023-08-24T16:44:57.6831647Z Configuring in ./gdb
...
2023-08-24T16:45:05.2050580Z checking for ELF support in BFD... no
...
```

From `gdb/configure` script in GDB source tree - it tries to compile a file with `"bfd.h"` and `"elf-bfd.h"` headers, so I would suspect missing Linux-specific dependencies included from those headers
```
# Add ELF support to GDB, but only if BFD includes ELF support.

  OLD_CFLAGS=$CFLAGS
  OLD_LDFLAGS=$LDFLAGS
  OLD_LIBS=$LIBS
  OLD_CC=$CC
  # Put the old CFLAGS/LDFLAGS last, in case the user's (C|LD)FLAGS
  # points somewhere with bfd, with -I/foo/lib and -L/foo/lib.  We
  # always want our bfd.
  CFLAGS="-I${srcdir}/../include -I../bfd -I${srcdir}/../bfd $CFLAGS"
  LDFLAGS="-L../bfd -L../libiberty"
  intl=`echo $LIBINTL | sed 's,${top_builddir}/,,g'`
  LIBS="-lbfd -liberty $intl $LIBS"
  CC="./libtool --quiet --mode=link $CC"
  { $as_echo "$as_me:${as_lineno-$LINENO}: checking for ELF support in BFD" >&5
$as_echo_n "checking for ELF support in BFD... " >&6; }
if ${gdb_cv_var_elf+:} false; then :
  $as_echo_n "(cached) " >&6
else
  cat confdefs.h - <<_ACEOF >conftest.$ac_ext
/* end confdefs.h.  */
#include <stdlib.h>
	   #include <string.h>
	   #include "bfd.h"
	   #include "elf-bfd.h"
int
main ()
{
return bfd_get_elf_phdr_upper_bound (NULL);

  ;
  return 0;
}
_ACEOF
if ac_fn_c_try_link "$LINENO"; then :
  gdb_cv_var_elf=yes
else
  gdb_cv_var_elf=no

fi
rm -f core conftest.err conftest.$ac_objext \
    conftest$ac_exeext conftest.$ac_ext

fi
{ $as_echo "$as_me:${as_lineno-$LINENO}: result: $gdb_cv_var_elf" >&5
$as_echo "$gdb_cv_var_elf" >&6; }
  CC=$OLD_CC
  CFLAGS=$OLD_CFLAGS
  LDFLAGS=$OLD_LDFLAGS
  LIBS=$OLD_LIBS
if test "$gdb_cv_var_elf" = yes; then
  CONFIG_OBS="$CONFIG_OBS elfread.o stap-probe.o dtrace-probe.o \
		gcore-elf.o elf-none-tdep.o"

$as_echo "#define HAVE_ELF 1" >>confdefs.h
```

ilg-ul, 2023-09-19:
Oops!

Thank you for your detailed bug report, I'll take a look during the next few days.

xobs, 2023-09-20:
Oh good, I was about to report something similar.

As reported by @zador-blood-stained , `12.3.0-1` was the first version that broke. `12.2.0-3` works just fine.

ilg-ul, 2023-09-20:
Both 12.3 and 13.2 use the same build scripts, so will be fixed together. 

I added a notice in the release notes to mention this issue.

ilg-ul, 2023-09-20:
I confirm the issue, it affects the arm toolchains too.

I added new tests when building gdb (#23), to catch this issue.

Builds with new releases for all toolchains are planned for the next few days.

ilg-ul, 2023-09-22:
I just published a new 12.3 release. Please check if it functional. 

13.2 will be updated soon.

zador-blood-stained, 2023-09-23:
Works fine for me
```
➜  test-gdb  % xpack-riscv-none-elf-gcc-12.3.0-2/bin/riscv-none-elf-gdb -q bme_sensor_oled.elf
Reading symbols from bme_sensor_oled.elf...
(gdb) disassemble main
Dump of assembler code for function main:
   0x000011c2 <+0>:     add     sp,sp,-16
   0x000011c4 <+2>:     sw      ra,12(sp)
   0x000011c6 <+4>:     sw      s0,8(sp)
   0x000011c8 <+6>:     li      a0,72
   0x000011cc <+10>:    auipc   ra,0x1ffff
...
```

ilg-ul, 2023-09-23:
Great, thank you for confirming.
I'll proceed with 13.2.

ilg-ul, 2023-09-25:
13.2 was published.

closed 2023-09-25 by ilg-ul