xpack-dev-tools/riscv-none-elf-gcc-xpack #28: __linux__ pragma

venkatmakam, 2024-01-28:
### Disclaimer

The xPack GNU RISC-V Embedded GCC is only a binary distribution of
the **SiFive RISC-V GNU Embedded Toolchain**,
and does not intend to add new functionality, or to fix existing problems.

If you have questions/suggestions related to the procedure used to
build/publish the binary distribution, continue reading the next sections.

If you have problems related to RISC-V GCC functionality, please use the
official [SiFive Forums](https://forums.sifive.com/).

### Prerequisites

Before entering a new ticket, please consider the following:

- [ ] read the documentation pages carefully (`share/doc/pdf/gcc/gcc.pdf`)
- [ ] check the [xPack Forums](https://www.tapatalk.com/groups/xpack/) for similar problems
- [ ] check the [GitHub Issues](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/)

If you still could not find an answer, if you have interesting use
cases, if you have custom configurations, and generally if you have
any experience that you would like to share with others, use the
[xPack Forums](https://www.tapatalk.com/groups/xpack/), the
GNU RISC-V Embedded GCC section.

If you are convinced that you identified a bug related to the binary
distribution, not to GNU RISC-V Embedded GCC itself (if you have doubts,
use the forum), or you have a pertinent suggestion how to enhance the
xPack GNU RISC-V Embedded GCC distribution, continue and register a new issue.

### Description
I am trying to cross compile a c program to risc-v linux environment. My code is structured for cross compiler and depends on the pragma __linux__ to know that the code is compiling for linux.

But with this compiler the pragma __linux__ is not defined. 
Do I need to define the pragma manually? what's the expected behavior for risc-v cross compilers.

### Versions

- [RISC-V GCC version] : 13.20.0.2
- [operating system] : windows 11

Please understand that without being able to reproduce the bug we cannot
identify your problem.

---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2024-01-28:
> Note: Remove unnecessary text after reading, before entering your ticket.

Did you read this notice?

venkatmakam, 2024-01-29:
> > Note: Remove unnecessary text after reading, before entering your ticket.
> 
> Did you read this notice?

sorry.. updated the issue description.

ilg-ul, 2024-01-29:
> I am trying to cross compile a c program to risc-v linux environment ... the pragma __ linux __ is not defined.

That's correct, the xPack GNU RISC-V **Embedded** GCC is for embedded targets (like bare metal), not for linux.

closed 2024-08-09 by ilg-ul