xpack-dev-tools/riscv-none-elf-gcc-xpack #29: Installation does not work.

greenscreenflicker, 2024-03-24:
### Description

I followed the installation instructions on the install section (https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack)
The code doesn't work.

### Steps to Reproduce

1. `xpm init`: No packages found.

2. `xpm install @xpack-dev-tools/riscv-none-elf-gcc@latest --verbose`: Package "@xpack-dev-tools/riscv-none-elf-gcc@latest" not found.
3. Tryed refereshing. Doesn't work either.
**Expected behaviour:** Should install

**Actual behaviour:** Doesn't install package.

### Versions

- [RISC-V GCC version]
- `lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy`

ilg-ul, 2024-03-24:
Are you behind a firewall/proxy?

greenscreenflicker, 2024-03-25:
I installed WSL2 on windows https://learn.microsoft.com/en-us/windows/wsl/install

I don't know,if there is any firewall. But apt under ubuntu wsl works.

See also: https://github.com/hathach/tinyusb/discussions/2525#discussioncomment-8895299

ilg-ul, 2024-03-25:
Wrong tool. See https://xpack.github.io to install the correct xpm.

greenscreenflicker, 2024-03-25:
And where exactly is the installtion tutorial?

greenscreenflicker, 2024-03-25:
install worked. i propose to put a direct link on the front side on the project, so that install instructions are more easy to find.

ilg-ul, 2024-03-25:
> put a direct link on the front side on the project

The existing one is not enough?

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack#prerequisites

closed 2024-08-09 by ilg-ul