xpack-dev-tools/riscv-none-elf-gcc-xpack #30: Once installed, how do I use it?

dioncullum29, 2024-04-22:
I performed the manual install of the library on my RHEL 8 machine (I downloaded and installed xpacks-riscv-none-elf-gcc-13.2.0-2-linux-x64.tar.gz), and I checked to make sure it was installed correctly, but I still am having a couple of issues.

I am attempting to build an existing project on our newer machines, and one of the issues this particular makefile is running into is there are a couple of GCC errors. The errors are:

> gcc: error: unrecognized argument in option '-mcmodel=medlow'
> gcc: error: unrecognized command line option '-mstack-bias'

Through my research, I was basically lead to believe that this was happening because I did not have riscv-unknown-elf-gcc as part of my PATH. This lead me to trying to get a version of this installed on our computers which lead me here. When following the install instructions, it specifically warns against adding this library to the system path, so I was wondering what my next step would be?

ilg-ul, 2024-04-22:
> and I checked to make sure it was installed correctly, 

What exactly did you check?

dioncullum29, 2024-04-22:
> What exactly did you check?

I followed the instructions on the install page which says:

> To check if the manually installed GCC starts, use something like:
> 
> $ ~/.local/xPacks/riscv-none-elf-gcc/xpack-riscv-none-elf-gcc-12.1.0-2/bin/riscv-none-elf-gcc --version
> riscv-none-elf-gcc (xPack GNU RISC-V Embedded GCC x86_64) 12.1.0

ilg-ul, 2024-04-22:
So, given this successful test, what is the correct path of the compiler executable?

dioncullum29, 2024-04-22:
Would it be "~/.local/xPacks/riscv-none-elf-gcc/xpack-riscv-none-elf-gcc-12.1.0-2/bin/riscv-none-elf-gcc" then?

ilg-ul, 2024-04-22:
That's correct.

dioncullum29, 2024-04-22:
Okay, so just to clarify, you are saying I have to now change the compiler path that I am using to this?

ilg-ul, 2024-04-22:
I'm saying that regardless where you install the compiler, you have to start it via a correct path, with the correct name.

You chose to unpack the archive manually, and you have to handle the path manually.

You can slightly automate things, and install the toolchain inside your project; in this case you get some links in the `<project>/xpacks/.bin` folder so you can start the compiler via shorter paths.

dioncullum29, 2024-04-22:
Okay, I understand what you mean now. Thank you so much for the help, and sorry if those were very basic and dumb questions. I am really new to all of this stuff and just working with a bunch of old projects on top of that. So thank you for bearing with me.

closed 2024-04-22 by dioncullum29

ilg-ul, 2024-04-22:
You're welcome!

Good luck with your projects.