xpack-dev-tools/riscv-none-elf-gcc-xpack #34: Make a new 14.2.0-2 release with binutils 2.43

ilg-ul, 2024-08-09:
https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/discussions/33

matsobdev, 2024-09-06:
New release fails to compile Pico 2 binary when `pico_riscv_gcc_zcb_zcmp`:

```
mateush@mateush:~/Dysk RAM/pico2$ cmake - S "~/Dysk RAM/pico2" -B "~/Dysk RAM/pico2/build"
PICO_SDK_PATH is /home/mateush/pico-sdk-2.0.0
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from /home/mateush/pico-sdk-2.0.0/src/boards/include/boards/pico2.h
Pico Platform (PICO_PLATFORM) is 'rp2350-riscv'.
Configuring toolchain based on PICO_COMPILER 'pico_riscv_gcc_zcb_zcmp'
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mateush/xpack-riscv-none-elf-gcc-14.2.0-2/bin/riscv-none-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mateush/xpack-riscv-none-elf-gcc-14.2.0-2/bin/riscv-none-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/mateush/xpack-riscv-none-elf-gcc-14.2.0-2/bin/riscv-none-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
TinyUSB available at /home/mateush/pico-sdk-2.0.0/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/mateush/pico-sdk-2.0.0/lib/btstack
cyw43-driver available at /home/mateush/pico-sdk-2.0.0/lib/cyw43-driver
lwIP available at /home/mateush/pico-sdk-2.0.0/lib/lwip
mbedtls available at /home/mateush/pico-sdk-2.0.0/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mateush/Dysk RAM/pico2/build
mateush@mateush:~/Dysk RAM/pico2$ cd build
mateush@mateush:~/Dysk RAM/pico2/build$ make -j 2
Scanning dependencies of target bs2_default
[  1%] Building ASM object pico-sdk/src/rp2350/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  2%] Linking ASM executable bs2_default.elf
[  2%] Built target bs2_default
Scanning dependencies of target bs2_default_padded_checksummed_asm
[  3%] Generating bs2_default.bin
[  5%] Generating bs2_default_padded_checksummed.S
[  5%] Built target bs2_default_padded_checksummed_asm
Scanning dependencies of target project_name
[  6%] Building C object CMakeFiles/project_name.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_stdlib/stdlib.c.obj
[  7%] Building C object CMakeFiles/project_name.dir/file_name.c.obj
[  9%] Building C object CMakeFiles/project_name.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_gpio/gpio.c.obj
/tmp/ccIIufYr.s: Assembler messages:
/tmp/ccIIufYr.s:1608: Error: unrecognized opcode `cm.mvsa01 s0,s1'
make[2]: *** [CMakeFiles/project_name.dir/build.make:108: CMakeFiles/project_name.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_gpio/gpio.c.obj] Błąd 1
make[2]: *** Oczekiwanie na niezakończone zadania....
make[1]: *** [CMakeFiles/Makefile2:2161: CMakeFiles/project_name.dir/all] Błąd 2
make: *** [Makefile:103: all] Błąd 2
mateush@mateush:~/Dysk RAM/pico2/build$
```

That seems important: ``/tmp/ccIIufYr.s:1608: Error: unrecognized opcode `cm.mvsa01 s0,s1'``

ilg-ul, 2024-09-06:
Hmmm... The 14.2.0-2 build uses the latest official gcc and the latest official binutils, I don't know what else to do to make it work. :-(

Frankly, I lost track of all these RISC-V extensions, and I don't know how well GCC supports them.

Do you have any suggestions?

ilg-ul, 2024-09-06:
@TommyMurphyTM1234, it looks like the Zcmp implementation still has some issues.

TommyMurphyTM1234, 2024-09-06:
> @TommyMurphyTM1234, it looks like the Zcmp implementation still has some issues.

Maybe @kito-cheng might be able to comment and clarify what version of Binutils supports the `Zcmp` extension `cm.mv*` instructions?

I can't see at a glance if/when this patch was integrated:

- https://sourceware.org/pipermail/binutils/2024-February/132263.html

> This patch supports Zcmp instruction 'cm.mva01s' and 'cm.mvsa01'.

ilg-ul, 2024-09-06:
I cannot see these changes in the 2.43.1 source either. So the latest binutils release includes only partial support for the `Zcmp` extension? 

@kito-cheng, can you confirm this?

matsobdev, 2024-09-06:
I'm fresh to all that RISC-V, so can not cast more light on it, but [CORE-V](https://www.embecosm.com/resources/tool-chain-downloads/#corev) compiler, 30 May 2024 does work. It is GCC 14.1.0. 22 Jun 2023 - GCC 12 something works as well. It's not any clue, but seems to be problems with `_zcmp` on a stable version, as [RPi forum](https://forums.raspberrypi.com/viewtopic.php?p=2244873#p2244873) says.

TommyMurphyTM1234, 2024-09-06:
> I'm fresh to all that RISC-V, so can not cast more light on it, but [CORE-V](https://www.embecosm.com/resources/tool-chain-downloads/#corev) compiler, 30 May 2024 does work. It is GCC 14.1.0. 22 Jun [...]

I think that the problem is clear. The patch to add support for the `Zcmp` extension's `cm.mv*` instructions to the Binutils assembler is on the `master` branch of Binutils but not in any released version up to and including 2.43.1. I presume that other toolchains that don't have a problem with these instructions are built with the necessary patch to Binutils.

ilg-ul, 2024-09-06:
> but [CORE-V](https://www.embecosm.com/resources/tool-chain-downloads/#corev) compiler, 30 May 2024 does work.

I did not check the build scripts, but I expect _top-of-tree_ to mean the very latest commits, not the stable releases.

kito-cheng, 2024-09-06:
> I cannot see these changes in the 2.43.1 source either. So the latest binutils release includes only partial support for the Zcmp extension?
>
> @kito-cheng, can you confirm this?

That's kinda epic mistake...we only merge partial zcmp support within 2.43.1...full support just landed last week, which means not included in any release yet.

[1] https://github.com/bminor/binutils-gdb/commit/ca2590d7804b4ea563eec6f1127ed17a00c30315
[2] https://sourceware.org/bugzilla/show_bug.cgi?id=32036

ilg-ul, 2024-09-06:
Thank you for clarifying this.

There will probably be a 14.2.0-3 after a new binutils release.

TommyMurphyTM1234, 2024-09-06:
So, I guess that the GCC 14 release notes change log is misleading? 

- https://gcc.gnu.org/gcc-14/changes.html

> - Support for the following standard extensions has been added:
>   - ...
>   - Code size reduction extensions:
>     - ...
>     - Zcmp
>     - ...

matsobdev, 2025-02-04:
binutils 2.44 are in town

ilg-ul, 2025-02-04:
We'll use it with GCC 15.