xpack-dev-tools/riscv-none-elf-gcc-xpack #37: MacOS arm64 toolchain - C++ link error: (.text._getentropy_r+0x12): undefined reference to `_getentropy'

MarcoBonino, 2024-10-11:
Hello!

I'm trying to use the toolchain 14.2.0 (also tried with 13.2.0). I'm always getting the same link error when my code use some c++ std things (like cout or vector).

I downloaded the toolchain but also tried to install it via package manager. Same error.

See the example:
```
$ cat main.cpp                                                                                                                                                                                                                               
#include <iostream>

int main() {
  std::cout << "Hello!" << std::endl;
  return 0;
}

```

Compilation:
```
$ /Users/makko/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/14.2.0-2.1/.content/bin/riscv-none-elf-g++ -o hello main.cpp 
/Users/makko/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/14.2.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/14.2.0/../../../../riscv-none-elf/bin/ld: /Users/makko/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/14.2.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/14.2.0/../../../../riscv-none-elf/lib/libc.a(libc_a-getentropyr.o): in function `_getentropy_r':
(.text._getentropy_r+0x12): undefined reference to `_getentropy'
collect2: error: ld returned 1 exit status
```

Am I doing something wrong ? Could you help on this ?

Thanks!

ilg-ul, 2024-10-11:
Assuming the link passed, what do you plan to do with the resulting binary?

MarcoBonino, 2024-10-11:
> Assuming the link passed, what do you plan to do with the resulting binary?

Run it! For now it will be on simulators, like spike, qemu...

ilg-ul, 2024-10-11:
> Run it! For now it will be on simulators, like spike, qemu...

And how do you expect the output stream resulting from writing to `cout` to reach you?

ilg-ul, 2024-10-11:
>  I could run and see the print using "spike pk my_binary".

That binary worked because that toolchain is specific to spike.

Generally you have to add code to redirect the output to something your environment supports, like semihosting, which is supported well by QEMU.

But things are not that obvious as you may think, you have to define a startup, linker scripts, etc.

You can try to instantiate a project from my [hello-world-qemu-template-xpack](https://github.com/micro-os-plus/hello-world-qemu-template-xpack) project template and take a look at the resulting code.