xpack-dev-tools/riscv-none-elf-gcc-xpack #38: Please reinstate the rv32ec/ilp32e libraries

basilhussain, 2024-11-17:
I note that all new releases since September 2024 (v14.2.0-2, v13.3.0-2, v12.4.0-2, v11.5.0-2) no longer feature rv32ec/ilp32e libraries. Was there a specific reason for the removal?

Such removal is detrimental to all (myself included) who rely on xPack RISC-V GCC to compile code for RV32E microcontrollers that support the 'C' compact instruction extension. For example, the very popular CH32V00x series of microcontrollers.

The impact is that any code being compiled and linked with `-mabi=ilp32e -march=rv32ec` gets _silently_ 'downgraded' to instead use the rv32e/ilp32e libraries, and the resulting compiled binaries become larger, even with _no other changes_. When this happens, there may be no clue readily apparent to the user as to why - I believe the only indication is via the library paths given in map file output (i.e. from `-Wl,-Map`).

Without the rv32ec library functions, compiled binary sizes can increase by hundreds of bytes or potentially kilobytes! When dealing with microcontrollers that have flash memory measured in only tens of kilobytes, such increases could make the difference between code fitting within the available flash and suddenly not.

Could you please reinstate the rv32ec/ilp32e libraries?

ilg-ul, 2024-11-17:
> Was there a specific reason for the removal?

I have nothing agains `rv32ec` in particular, just that the previous list of libraries was combinatorial and grew too large. Having no other inspiration, when I had to cut it to a reasonable size, I used the same list as Debian packages.

> Could you please reinstate the rv32ec/ilp32e libraries?

I'll consider this for the next release.

I would like to ask you a favour. I have no experience with the new RISC-V devices available on the market. Could you do a small study and check if their architectures are supported by the xPack toolchain, and possibly suggest new libraries to be included?

ilg-ul, 2024-11-17:
I took a look at the current list. As a temporary workaround, if your code does not use atomic instructions, could you use the `rv32eac` library?

basilhussain, 2024-11-17:
> I'll consider this for the next release.

Thanks! 😃

> I would like to ask you a favour. I have no experience with the new RISC-V devices available on the market. Could you do a small study and check if their architectures are supported by the xPack toolchain, and possibly suggest new libraries to be included?

I'm not really familiar with the variety of what's around, but I found [this article](https://muxup.com/2023q1/commercially-available-risc-v-silicon) which seems to give a pretty good run-down.

For WCH microcontrollers in particular (which is what I'm most familiar with), while device models are too numerous to list here, they basically only have three different CPU cores:

* QingKeV2A - RV32EC
* QingKeV2C - RV32EmC*
* QingKeV3A - RV32IMAC
* QingKeV3B - RV32IMCB*
* QingKeV3C - RV32IMCB*
* QingKeV4A - RV32IMAC
* QingKeV4B - RV32IMAC
* QingKeV4C - RV32IMAC
* QingKeV4F - RV32IMACF

(* 'm' - Zmmul only; 'B' - unsure if conforms to ratified standard)

As far as microcontrollers goes in general - as opposed to larger SoCs - I think it's safe to say that at least the following is needed:

* EC
* IMC
* IMAC
* IMACF

RISC-V microcontrollers pretty much always support 'C', because small code size is typically of great importance in embedded applications, due to limited on-chip flash sizes.

> As a temporary workaround, if your code does not use atomic instructions, could you use the rv32eac library?

If it means the library code will contain Atomic extension instructions, then no. CH32V00x is strictly RV32EC only; I suspect you'd just get an illegal instruction hard fault if it tried to execute any 'A' instruction.

ilg-ul, 2024-11-17:
> but I found [this article](https://muxup.com/2023q1/commercially-available-risc-v-silicon)

Yes, good article.

> I think it's safe to say that at least the following is needed...

`imacf` is present as `rv32imafc_zicsr`.

So only `ec`, `imc`, `imac` are missing?

Tommy @TommyMurphyTM1234, any comments on this proposal?

> If it means the library code will contain Atomic extension instructions, then no. CH32V00x is strictly RV32EC only; I suspect you'd just get an illegal instruction hard fault if it tried to execute any 'A' instruction.

I did not check the compiler and the library source code, so I might be wrong, but I don't think that the compiler has any reasons to generates atomic instructions for usual C code, you need to explicitly include <stdatomic.h> to get atomic support, which I doubt you do.

I suggest you give it a try.

It'll probably take me some time, possibly a few weeks, to prepare a new maintenance release with the new libraries.

basilhussain, 2024-11-17:
> So only `ec`, `imc`, `imac` are missing?

That would appear to be the case, yes. I would bet that set of 4 would cover 99% of RISC-V microcontrollers.

> I did not check the compiler and the library source code, so I might be wrong, but I don't think that the compiler has any reasons to generates atomic instructions for usual C code, you need to [...]

I would hope that would be the case, but I wouldn't feel safe relying on that assumption.

In any case, I tried compiling one of my code bases with 13.3.0-2 and rv32eac and I only get a single-byte difference between that and the same compiled with 13.3.0-1 and rv32ec. So, for this code base at least (which uses minimal libgcc and libc functions), I get usable results.

(Side note: I'm a bit confused by this single-byte difference, as it's in a section of my own code, and it's a `addi a3,a3,-1316` that has become a `addi a3,a3,-1315`. No idea why an offset literal value would change when literally every other byte in the binary is identical... 😕)

> It'll probably take me some time, possibly a few weeks, to prepare a new maintenance release with the new libraries.

Not a problem for me personally. I'm just continuing to use the older versions anyway. 😄

For people who've inadvertently upgraded their riscv-none-elf-gcc package, in the meantime they could downgrade I suppose.

ilg-ul, 2024-11-17:
To summarise, the missing libraries are:

- rv32ec
- rv32imc
- rv32imac

The WCH specific extensions are out of my scope.

TommyMurphyTM1234, 2024-11-18:
> The WCH specific extensions are out of my scope.

Can you clarify what you're referring to here? 
I presume something like...

1. [`__attribute __ ((interrupt("WCH-Interrupt-fast")))`](https://www.reddit.com/r/RISCV/comments/126262j/notes_on_wch_fast_interrupts/)
2. WCH (or QingKe?) proprietory non-standard extension [`xw`](https://stackoverflow.com/questions/76335036/gnu-risc-v-embedded-gcc-throws-x-isa-extension-xw-must-be-set-with-the-versio)
3. Other?

ilg-ul, 2024-11-18:
All of the above.

basilhussain, 2024-11-18:
> The WCH specific extensions are out of my scope.

Yes, of course, because none of that is in mainline GCC. Although it would be nice if they added XW support.

ilg-ul, 2024-11-18:
> it would be nice if they added XW support.

Sure, ask them, perhaps they'll do it. I have no contacts there.

TommyMurphyTM1234, 2024-11-18:
> > The WCH specific extensions are out of my scope.
> 
> Yes, of course, because none of that is in mainline GCC. Although it would be nice if they added XW support.

Did MounRiver ever release their modifications to GCC that added support for it, as they're obviously obliged to do under GPL?

ilg-ul, 2024-11-18:
No idea.

cnlohr, 2024-11-18:
Also - somewhat unrelated - have you considered separating out RV32/RV64 as different packages? I will admit the 500MB download was quite the chonker, compared to sysprog's RV32 compiler zip (which even bundles Make) at an 82MB zip.

I don't have a strong preference, was just curious.

ilg-ul, 2024-11-18:
> have you considered separating out RV32/RV64 as different packages

Only as a last resort.

> 500MB download 

Now the archives are in the 300MB range. Still large, but somehow manageable.

cnlohr, 2024-11-20:
All fair.  Just 🙏 for the rv32ec/ilp32e.

cnlohr, 2024-12-04:
Just wondering, what is the timeframe for expected inclusion?  I would really like to switch the target for ch32v003fun to xpack, from sysprogs, since they are stuck back at GCC10.

ilg-ul, 2024-12-04:
I'm really busy these days with the new web for the whole xPack project. It might take 1-2 more weeks.

TommyMurphyTM1234, 2024-12-04:
> To summarise, the missing libraries are:
> 
> * rv32ec
> * rv32imc
> * rv32imac
> 
> The WCH specific extensions are out of my scope.

If it helps at all I can prepare a PR for this.

If so then maybe I can get some clarification.

(1) I presume that `rv32ec-ilp32e--` should be added? Should it have any additional aliasing - e.g. `rv32ec-ilp32e--m` - or not?

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/89586bd9923f53f3a8070e3a3dad7e5f1c76a5d1/build-assets/scripts/versioning.sh#L193-L198

(2) `rv32imc` is listed above as missing but isn't it already covered by this (below) or should that be replaced by the explicit `rv32imc-ilp32--`?

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/89586bd9923f53f3a8070e3a3dad7e5f1c76a5d1/build-assets/scripts/versioning.sh#L202

(3) `rv32imac` is listed above as missing but this seems to be already catered for?

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/89586bd9923f53f3a8070e3a3dad7e5f1c76a5d1/build-assets/scripts/versioning.sh#L207

(4) Should anything be change/be added here?

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/89586bd9923f53f3a8070e3a3dad7e5f1c76a5d1/build-assets/scripts/versioning.sh#L229-L233

(5) If making changes I would endeavour to update the `multilib-generator` comment and any other relevant comments:

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/89586bd9923f53f3a8070e3a3dad7e5f1c76a5d1/build-assets/scripts/versioning.sh#L190
- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/89586bd9923f53f3a8070e3a3dad7e5f1c76a5d1/build-assets/scripts/versioning.sh#L144-L150
- etc.

ilg-ul, 2024-12-04:
The problem is not adding three lines in the script, but the many hours that the builds take.

TommyMurphyTM1234, 2024-12-04:
> The problem is not adding three lines in the script, but the many hours that the builds take.

It would still help to clearly identify the changes that need to be made.
They are not clear to me from the previous discussion.

ilg-ul, 2024-12-04:
> I would really like to switch the target for ch32v003fun to xpack

What platform are using? I can make a pre-release these days so you can try it.

ilg-ul, 2024-12-04:
> It would still help to clearly identify the changes that need to be made.

I updated the build script. There were only two changes needed, the other two combinations were already there.

- `rv32e-ilp32e--c` was changed to `rv32e-ilp32e--` and `rv32ec-ilp32e--`
- `rv32im-ilp32--c` was changed to `rv32im-ilp32--` and `rv32imc-ilp32--`

The meaning of `--c` is that the given combination (like `rv32e`) is used for both `rv32e` and `rv32ec`. In other words, the library is compiled with `rv32e` (without `c`) and is linked to applications compiled with or without `c`. The result is perfectly functional, but the libraries, not using the compressed instructions, are slightly larger, as @basilhussain noticed.

ilg-ul, 2024-12-04:
BTW @cnlohr, did you try the existing binaries and you could not build the binary due to lack of space in flash?

cnlohr, 2024-12-04:
Just for background, I'm the lead maintainer of https://github.com/cnlohr/ch32v003fun - for projects like the rv003usb bootloader where the USB stack has to fit into the 1920-byte bootloader, the C extension is absolutely required.

If you have a test build ready without the extra `ilp32e--c`, and instead just `ilp32e` I can definitely test it out!

I am personally unaware of any differences at all between ilp32e and ilp32e--c or between `rv32e-ilp32e--` / `rv32ec-ilp32e--` but, the core `ilp32ec` is a pretty hard requirement for us to use it... And, there are massive differences between `ilp32` and `ilp32e`

ilg-ul, 2024-12-04:
> If you have a test build ready without the extra ilp32e--c, and instead just ilp32e I can definitely test it out!

I'll have a darwin-x64 archive probably in a few hours.

TommyMurphyTM1234, 2024-12-04:
> I am personally unaware of any differences at all between ilp32e and ilp32e--c or between `rv32e-ilp32e--` / `rv32ec-ilp32e--` but, the core `ilp32ec` is a pretty hard requirement for us to use it... And, there are massive differences between `ilp32` and `ilp32e`

I'm a bit confused...

I thought that ch32v003fun was completely standalone and didn't link any startup or library code from the toolchain but provided everything needed to produce the linked executable ELF files?

If that was the case then it would make no difference what multilibs (and startup code) were bundled with the toolchain given that they would be ignored. 

And regardless of the configuration of the toolchain and multilibs any toolchain can generate code for `rv32ec/ilp32e` once `-march=rv32ec -mabi=ilp32e` are passed at compile/link time.

cnlohr, 2024-12-04:
ch32v003fun typically does not link any external things, but on parts like the ch32v208 (not ilp32e admittedly) there's only a binary blob for the bluetooth.

One other thing to consider is the libgcc stuff needs to be for ilp32e, i.e. for multiplying numbers, etc.  It needs to be able to call the libgcc backend functions.  So I would expect ilp32e is needed?  (Sorry, I am not an expert on this part).

ilg-ul, 2024-12-04:
> And regardless of the configuration of the toolchain and multilibs any toolchain can generate code for rv32ec/ilp32e once -march=rv32ec -mabi=ilp32e are passed at compile/link time.

That's correct.

A standalone project with so tight memory constraints should probably avoid all library calls. I took a quick look at your project, and apparently it defines locally lots of C standard functions, so it should be fine.

However I have to note that it might be possible for GCC to call some internal functions; this can be clarified in the linker map. If the linker does not include such functions for your application, then there should be no difference between using a toolchain built with explicit `rv32ec-ilp32e--` and one built with `rv32e-ilp32e--c`. 

The only difference between the toolchains is that with the explicit two `rv32e-ilp32e--` & `rv32ec-ilp32e--` definition there are two versions of the system libraries, while with the single `rv32e-ilp32e--c` there is only one, without compression, which can be used very well regardless how the application was compiled, with or without compression.

ilg-ul, 2024-12-04:
> the libgcc stuff needs to be for ilp32e, i.e. for multiplying numbers, etc. 

Yes, this is the internal functions that I mentioned before.  in the current release these functions are not compiled with compression. I am curious how much difference makes these functions.

> So I would expect ilp32e is needed?

Yes, if your device has only 16 registers, ilp32e is needed.

cnlohr, 2024-12-04:
Does that mean that the difference between `rv32ec-ilp32e--` and `rv32e-ilp32e--c` would be the generation of `__mulsi3`?  I mean don't get me wrong, it would be pretty baller to get -c versions of `__mulsi3`.

TommyMurphyTM1234, 2024-12-04:
> ch32v003fun typically does not link any external things, but on parts like the ch32v208 (not ilp32e admittedly) there's only a binary blob for the bluetooth.
>
> One other thing to consider is the [...]

Ah, ok, that makes sense. My confusion stemmed from me only looking at some basic self-contained examples that didn't link any toolchain provided startup code or libraries. 

As far as I know any `rv32e*` architecture requires the `ilp32e` ABI.

In case it clarifies at all, when a multilib is configured and built for `rv32e-ilp32e--c` it means that the code generated will not use any compressed instructions but if `-march=rv32ec-mabi=ilp32e` is specified at compile/link time then the `rv32e/ilp32e` will be linked (if there is no multilib specifically for `rv32ec/ilp32e`) so the results will be suboptimal. I.e. the target supports the C extension but the libraries don't take advantage of that.

cnlohr, 2024-12-04:
Reading you message @TommyMurphyTM1234 -- I am still a little confused, does this have anything to do with the generation of functions like `__mulsi3`?  I think you're saying it does.

Also - also @ilg-ul this is not urgent for us.  Our users have a workable solution at the moment. But, I will have time this weekend, and would love to further streamline the development experience. Using xpack binaries would be just so cool.

ilg-ul, 2024-12-04:
> Does that mean that the difference between rv32ec-ilp32e-- and rv32e-ilp32e--c would be the generation of code like __mulsi3? 

Yes

> I mean don't get me wrong, it would be pretty baller to get -c versions of __mulsi3. Just not required.

The next release will have -c versions of all libgcc.

I suggest you try a first build with the existing 14.2.0-2 and later compare with 14.2.0-3 which has the compressed libgcc code.

TommyMurphyTM1234, 2024-12-04:
> Reading you message @TommyMurphyTM1234 -- I am still a little confused, does this have anything to do with the generation of functions like `__mulsi3`? I think you're saying it does.

Sorry, I don't know offhand and would have to look at the disassembly of a multilib for `rv32e-ilp32e--c` versus one for `rv32ec-ilp32e--`.

> Also - also @ilg-ul this is not urgent for us. Our users have a workable solution at the moment. 

Do you mean the de facto use of the SysProgs GCC 10 for RISC-V bare-metal or an earlier version of the xPack GCC for RISC-V than the September releases that omitted `rv32ec/ilp32e` multilib support?

cnlohr, 2024-12-04:
@ilg-ul can I give you money or a coffee or something for wasting your time? Two people just said that 
xpack-riscv-none-elf-gcc-14.2.0-2-win32-x64 didn't work.  But I just tested, and it worked fine.

ilg-ul, 2024-12-04:
> an earlier version of the xPack

Ah, right, 14.2.0-1 should also have support for `rv32ec-ilp32e--`.

cnlohr, 2024-12-04:
To be fair what was discussed in this thread would be great... But it does work... And the output is better, off the bat than the SysProgs GCC 10.

I would be happy to compare.  Either way, I will get started this weekend with trying to find a good experience using the sysprogs GCC.  Is there anything that would stop me from wrapping it in an installer, license or etiquette wise?

TommyMurphyTM1234, 2024-12-04:
> @ilg-ul can I give you money or a coffee or something for wasting your time? Two people just said that xpack-riscv-none-elf-gcc-14.2.0-2-win32-x64 didn't work. But I just tested, and it worked fine.

Maybe it worked but the resulting code isn't optimal because the linked libraries doesn't exploit the compressed extension instructions?

ilg-ul, 2024-12-04:
If you want to support the project, please see:

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/getting-started/#enjoyed-using-this-project-let-us-know

ilg-ul, 2024-12-04:
> Is there anything that would stop me from wrapping it in an installer, license or etiquette wise?

As long as you respect the MIT license terms, you're fine.

But I suggest you wait for the final new release of 14.2.0-3.

cnlohr, 2024-12-04:
Done! Thank you for the donation link.

Just wondering... This is probably a question for @TommyMurphyTM1234 but if I wanted to compare code output between ipl32e--c and ilp32e, how would I go about that?

Would definitely be respecting the MIT license. 

Oddly enough, even with xpack-riscv-none-elf-gcc-14.2.0-2, I am getting rv32ec code gen for libgcc functions somehow?

```
000000a0 <__mulsi3>:
  a0:	862a                	mv	a2,a0
  a2:	4501                	li	a0,0
  a4:	0015f693          	andi	a3,a1,1
  a8:	c291                	beqz	a3,ac <__mulsi3+0xc>
  aa:	9532                	add	a0,a0,a2
  ac:	8185                	srli	a1,a1,0x1
  ae:	0606                	slli	a2,a2,0x1
  b0:	f9f5                	bnez	a1,a4 <__mulsi3+0x4>
  b2:	8082                	ret
  b4:	0000                	unimp
	...
```

I am VERY excited to make the primary path for our users xpack, instead of sysprogs.

cnlohr, 2024-12-04:
That said, I will definitely wait for 14.2.0-3 to do any bundling.  I was planning on using the NSIS installer like I did for the tinycc installer here: https://github.com/cnlohr/tinycc-win64-installer

cnlohr, 2024-12-04:
Also - I am **very confused** right now. I don't understand how I am getting EC libgcc, because it's not in riscv-elf-none/14.2.0-2 
[image]

cnlohr, 2024-12-04:
And, I looked in rv32e's libgcc.a, and it only has rv32e, not rv32ec mulsi3.  I can't figure out where the rv32ec mulsi3 is coming from.

```
00000000 <__mulsi3>:
   0:	00050613          	mv	a2,a0
   4:	00000513          	li	a0,0

00000008 <.L1>:
   8:	0015f693          	andi	a3,a1,1
   c:	00068463          	beqz	a3,14 <.L2>
  10:	00c50533          	add	a0,a0,a2

00000014 <.L2>:
  14:	0015d593          	srli	a1,a1,0x1
  18:	00161613          	slli	a2,a2,0x1
  1c:	fe0596e3          	bnez	a1,8 <.L1>
  20:	00008067          	ret
  ```

TommyMurphyTM1234, 2024-12-04:
> And, I looked in rv32e's libgcc.a, and it only has rv32e, not rv32ec mulsi3. I can't figure out where the rv32ec mulsi3 is coming from.
>
> ```
> 00000000 <__mulsi3>:
>    0:	00050613          	mv	a2,a0 [...]

Are functions like `__mulsi3` generated by the compiler rather than linked from the library? So when you compile with `-march=rv32ec -mabi=ilp32e` that causes them to be generated using compressed instructions?

ilg-ul, 2024-12-04:
I don't know, but I would guess that you're messing something.

ilg-ul, 2024-12-04:
>  generated by the compiler rather than linked from the library?

Enable the linker map and look there.

TommyMurphyTM1234, 2024-12-04:
> Just wondering... This is probably a question for @TommyMurphyTM1234 but if I wanted to compare code output between ipl32e--c and ilp32e, how would I go about that?

I would be inclined (and intend) to build the `riscv-gnu-toolchain` in two variants (below) and then compile some of the ch32v003fun examples and look at the disassembly listings. I will probably do this tomorrow unless everything becomes clearer before then. :-)
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

./configure --prefix=`pwd`/rv32e-toolchain --with-multilib-generator="rv32e-ilp32e--c"
make 

./configure --prefix=`pwd`/rv32ec-toolchain --with-multilib-generator="rv32ec-ilp32e--"
# make distclean?
make
```

cnlohr, 2024-12-04:
I am absolutely not set up for that at this time. I would rather trust the direction you guys are headed, and go with 14.2.0-3 when it's released.

That said, I am happy to compare -2 and -3.  but -1 and -2 produce identical code.  I really wish I could remember who pointed me to this issue when they said there were issues so I could see what issues they were seeing.

ilg-ul, 2024-12-04:
> -1 and -2 produce identical code.

That's a bit surprising, but I don't have time to investigate now.

TommyMurphyTM1234, 2024-12-04:
> @TommyMurphyTM1234 I am absolutely not set up for that at this time. I would rather trust the direction you guys are headed, and go with 14.2.0-3 when it's released.

Just to clarify - my suggested `riscv-gnu-toolchain` experimentations are separate from what @ilg-ul is doing with xPack RISC-V GCC but should still help to clarify the precise nature of the code generated/linked in the two cases (`rv32e-ilp32e--c` and `rv32ec-ilp32e--`).

ilg-ul, 2024-12-04:
I might be wrong, but the surprising result might be explained by using `-flto`.

cnlohr, 2024-12-04:
no worries! I am just excited to move on and leave sysprogs behind!

cnlohr, 2024-12-04:
It "looks like" it might be linking to the libgcc.a in lib\gcc\riscv-none-elf\14.2.0 - not in any of the arch-specific folders.  Which has mulsi3, but, oddly enough other functions in there seem to assume rv32imac

cnlohr, 2024-12-04:
Ok, yeah, that's scary, that it happens to be dumb luck that everything works in the cases I tested.  I am very excited to get -3!

ilg-ul, 2024-12-05:
Please test the pre-release binaries:

- https://github.com/xpack-dev-tools/pre-releases/releases/tag/test

ilg-ul, 2024-12-05:
> It "looks like" it might be linking to the libgcc.a in lib\gcc\riscv-none-elf\14.2.0 - not in any of the arch-specific folders.

There is no _looks like_. and there is no magic, the behaviour is deterministic and traceable. 

The `.map` files list each external file with the full path of the library. For example:

```
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/13.2.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/13.2.0/rv64imafdc_zicsr/lp64d/libgcc.a(trunctfdf2.o)
```

Search for `libgcc` in your map file.

cnlohr, 2024-12-05:
Aaaha! Sorry, I have always used objdump to make my map files, it is indeed better to do map file creation by the linker.

It turns out it's using a vendored libgcc!  Thanks for having me check that. 

If I don't use the vendored gcc, it does still works with -2, but, the flash usage goes back up, since it is using the non-c libgcc.

Also, I tested your pre-release binaries and they work perfectly.  I will base my install off of them.

Thank you so much for your effort on this.

TommyMurphyTM1234, 2024-12-05:
> It "looks like" it might be linking to the libgcc.a in lib\gcc\riscv-none-elf\14.2.0 - not in any of the arch-specific folders. Which has mulsi3, but, oddly enough other functions in there seem to assume rv32imac

FYI...

If a specific multilib cannot be matched for the `arch/abi` selected at compile/link time (taking account of any "reuse" patterns if applicable) then the linker will fall back on linking the toolchain default libs. And in the case of the xPack toolchain that's `rv32imac/ilp32`.

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/blog/2024/09/06/riscv-none-elf-gcc-v14-2-0-2-released/#supported-libraries

Since this uses the `ilp32` ABI it presumably won't work currently on an `rv32e` target which requires `ilp32e`?

TommyMurphyTM1234, 2024-12-05:
> It turns out it's using a vendored libgcc! Thanks for having me check that.
> 
> If I don't use the vendored gcc, it does still works with -2, but, the flash usage goes back up, since it is using the non-c libgcc.

Sorry, what exactly do you mean by "vendored" GCC/libgcc here?

[25 more replies]

closed 2024-12-06 by ilg-ul