xpack-dev-tools/riscv-none-elf-gcc-xpack #43: std::log2(x) is not a member of 'std'

yy172, 2025-06-17:
When I compile the statement mentioned in the title using GCC 12.3.0-2, the compilation fails with the error: "error: 'log2' is not a member of 'std'; did you mean 'log'?" However, when using GCC 12.4.0-2, it compiles successfully, with identical compilation parameters and source code. The only difference is the release version of GCC 12. Could this issue be caused by the parameters configured when building GCC 12 from source? If so, specifically which parameter might be responsible?

ilg-ul, 2025-06-17:
Frankly, I don't know what caused this. It may be the build configuration, but it may also be the source code.

yy172, 2025-06-17:
Okay, I was able to compile successfully using riscv-none-elf-gcc 12.3 on an online compiler website. This suggests the issue might indeed be with my local build configuration. However, I couldn't spot any obvious problems when reviewing the GCC configuration parameters. I might try replacing the source code to verify this assumption through reverse validation.

TommyMurphyTM1234, 2025-06-17:
No self-contained reproducible test case/instructions?

yy172, 2025-06-19:
There is a test case that uses std::log2, and it compiled without errors. According to the log, this case was compiled using   "g++ -xbb"