xpack-dev-tools/riscv-none-elf-gcc-xpack #44: Will the following arch/abi combinations be added in the future

helloeagleyang, 2025-08-05:
rv32imac_zicsr_zifencei_za_ab_zc_cs/ilp32
rv32imafc_zicsr_zifencei_za_ab_zc_cs/ilp32f
rv32imafdc_zicsr_zifencei_za_ab_zc_cs/ilp32d

helloeagleyang, 2025-08-06:
> What devices implement them?

HPM6200, HPM6E00, HPM6P00, and HPM5E00 series support them. here is the website:www.hpmicro.com

Both are based on RV32GCB

helloeagleyang, 2025-08-07:
> Are you sure that **these** are correct?
> 
> > rv32imac_zicsr_zifencei_**za_ab_zc_cs**/ilp32
> > rv32imafc_zicsr_zifencei_**za_ab_zc_cs**/ilp32f
> > rv32imafdc_zicsr_zifencei_**za_ab_zc_cs**/ilp32d
> 
> > All are based on RV32GCB
> 
> Doesn't the B extension encompass the following?
> 
> * `Zba`
> * `Zbb`
> * `Zbs`
> * https://github.com/riscv/riscv-b

Sorry for my typo. compared to the B extension, the MCU series mentioned above support zba, zbb, zbc, zbs.

Here are the multilib configured on my side

```code
--enable-multilib --with-multilib-generator='rv32imac_zicsr_zifencei-ilp32--;rv32imafc_zifencei-ilp32--;rv32imafc_zifencei-ilp32f--;rv32imafdc-ilp32--;rv32imafdc-ilp32f--;rv32imafdc-ilp32d--;rv32imac_zicsr_zifencei_zba_zbb_zbc_zbs-ilp32--;rv32imafc_zifencei_zba_zbb_zbc_zbs-ilp32--;rv32imafc_zifencei_zba_zbb_zbc_zbs-ilp32f--;rv32imafdc_zba_zbb_zbc_zbs-ilp32--;rv32imafdc_zba_zbb_zbc_zbs-ilp32f--;rv32imafdc_zba_zbb_zbc_zbs-ilp32d--' --with-abi=ilp32 --with-arch=rv32imac_zicsr_zifencei --with-tune=rocket --with-isa-spec=20191213 
```

ilg-ul, 2025-08-07:
> Yes, they are required to build the libraries at least on my side

I see. Not good.

I currently don't know how to handle this. For these 3 libraries we might hardly accept an exception, but for the other we'll most likely use common libraries without these extensions.

I'm not familiar with these extensions, all I know is that they have to do with bit manipulations. What is the reason for using them? Code size? Speed? Do bit manipulations really make a difference inside the system libraries?

And what is the `cs` exception?

> I tested the Ubuntu 22.04 and Ubuntu 24.04, both can support the extensions

Ok, I'll check the build configurations from the latest Ubuntu, perhaps I can get some ideas what would be a better combination of multi-libs.

helloeagleyang, 2025-08-07:
> What is the reason for using them? Code size? Speed?

The B extension can achieve both the code size reduction and performance enhancement for the Bit related operations, we see it help improve the networking related throughput about 50% on the HPM5E00 series. 

> Do bit manipulations really make a difference inside the system libraries?

From the library's perspective, I am not sure about the  benefit.

> And what is the cs exception?

Zbs: Single-bit instructions, See more information in Section 1.5. from https://five-embeddev.com/riscv-bitmanip/1.0.0/bitmanip.html

ilg-ul, 2025-08-07:
So the `-cs` in the initial post was a typo?

TommyMurphyTM1234, 2025-08-07:
> Yes, I mean zba, zbb, zbc, zbs, considering the generic case, maybe we can discard the zbc extension as it is not in official B extension set in the ISA spec.

Just to be clear, `Zbc` is not part of the `B` extension which encompasses just the `Zba`, `Zbb` and `Zbs` sub(?) extensions.

> Are these extensions supported by the latest Debian cross compiler?

All of these extensions have been supported by GCC since version 12:

- https://gcc.gnu.org/gcc-12/changes.html#riscv

Unfortunately RISC-V International seem to make it very difficult to find the authoritative information about extensions, their specifications and their status because they regularly change the layouts of the relevant sites and leave lots of legacy non-authoritative information and broken links online. 

E.g. their specifications page below lists the different bit manipulation extensions as ratified extension but doesn't link to the ratified specification documentation or GitGub/other pages (maybe the specifications for such extensions are integrated into the main ISA or Privileged specs once ratified?); gives the impression that `Zba`, `Zbb`, `Zbs` and `Zbc` are linked when `Zbc` is a separate extension; and the Jira link on the GitHub page about the `B` extension that I posted earlier gives a 404...

- https://riscv.org/specifications/ratified/
  - https://lf-riscv.atlassian.net/wiki/spaces/HOME/pages/16154732/Ratified+Extensions
- https://github.com/riscv/riscv-b
  - https://lf-riscv.atlassian.net/jira/browse/RVG-122